timvideos/HDMI2USB-jahanzeb-firmware #105: Debug unreliable screen detection and possible EDID issue

mithro, 2015-01-16:
Currently we seem to randomly get issues with the Atlys board not being detected.

mithro, 2015-01-16:
Useful debugging information from David Airlie
```
echo 6 > /sys/module/drm/parameters/debug

Documentation/EDID
drivers/gpu/drm/i915/intel_i2c.c
drivers/gpu/drm/drm_edid.c
```

https://01.org/linuxgraphics/documentation/2013-intel-core-processor-family

mithro, 2015-01-16:
@shenki @ajit

mithro, 2015-01-16:
http://stackoverflow.com/questions/5065159/reading-edid-from-eeprom

mithro, 2015-01-16:
Example of the EDID shown in xrandr output different despite EDID blocks being the same;
```
HDMI1 connected 1024x768+1600+0 (0xb3) normal (normal left inverted right x axis y axis) 0mm x 0mm
	Identifier: 0x45
	Timestamp:  1800325528
	Subpixel:   unknown
	Gamma:      1.0:1.0:1.0
	Brightness: 1.0
	Clones:    
	CRTC:       1
	CRTCs:      0 1 2
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	EDID: 
		00ffffffffffff00283a000001000000
		06170103810000780f5ec0a4594a9825
		20505400080061400101010101010101
		01010101010164190040410026301888
		3600000000000018000000fc0048444d
		49325553420a0a0a0a0a000000fe004a
		4148414e5a454241484d414400000010
		000020300808880000000000001e007d
	Broadcast RGB: Automatic 
		supported: Automatic, Full, Limited 16:235
	audio: auto 
		supported: force-dvi, off, auto, on
  1024x768 (0xb3)   65.0MHz -HSync -VSync *current +preferred
        h: width  1024 start 1048 end 1184 total 1344 skew    0 clock   48.4KHz
        v: height  768 start  771 end  777 total  806           clock   60.0Hz
```
-----
```
HDMI1 connected 1024x768+1600+0 (0xb3) normal (normal left inverted right x axis y axis) 0mm x 0mm
	Identifier: 0x45
	Timestamp:  1800395832
	Subpixel:   unknown
	Gamma:      1.0:1.0:1.0
	Brightness: 1.0
	Clones:    
	CRTC:       1
	CRTCs:      0 1 2
	Transform:  1.000000 0.000000 0.000000
	            0.000000 1.000000 0.000000
	            0.000000 0.000000 1.000000
	           filter: 
	EDID: 
		00ffffffffffff00283a000001000000
		06170103810000780f5ec0a4594a9825
		20505400080061400101010101010101
		01010101010164190040410026301888
		3600000000000018000000fc0048444d
		49325553420a0a0a0a0a000000fe004a
		4148414e5a454241484d414400000010
		000020300808880000000000001e007d
	Broadcast RGB: Automatic 
		supported: Automatic, Full, Limited 16:235
	audio: auto 
		supported: force-dvi, off, auto, on
  1024x768 (0xb3)   65.0MHz -HSync -VSync *current +preferred
        h: width  1024 start 1048 end 1184 total 1344 skew    0 clock   48.4KHz
        v: height  768 start  771 end  777 total  806           clock   60.0Hz
  800x600 (0xb4)   40.0MHz +HSync +VSync
        h: width   800 start  840 end  968 total 1056 skew    0 clock   37.9KHz
        v: height  600 start  601 end  605 total  628           clock   60.3Hz
  800x600 (0xb5)   36.0MHz +HSync +VSync
        h: width   800 start  824 end  896 total 1024 skew    0 clock   35.2KHz
        v: height  600 start  601 end  603 total  625           clock   56.2Hz
  848x480 (0x2d4c)   33.8MHz +HSync +VSync
        h: width   848 start  864 end  976 total 1088 skew    0 clock   31.0KHz
        v: height  480 start  486 end  494 total  517           clock   60.0Hz
  640x480 (0x2d4d)   25.2MHz -HSync -VSync
        h: width   640 start  656 end  752 total  800 skew    0 clock   31.5KHz
        v: height  480 start  489 end  492 total  525           clock   59.9Hz
```

mithro, 2015-01-16:
Logs from connecting a HDMI2USB with no firmware loaded;

```
[2003680.341651] [drm:intel_crtc_cursor_set], cursor off
[2003683.484459] [drm:intel_crtc_cursor_set], cursor off
[2003687.986194] [drm:intel_hpd_irq_handler], Received HPD interrupt on PIN 4 - cnt: 0
[2003687.986217] [drm:i915_hotplug_work_func], running encoder hotplug functions
[2003687.986221] [drm:i915_hotplug_work_func], Connector HDMI-A-1 (pin 4) received hotplug event.
[2003687.986223] [drm:i915_hotplug_work_func], Connector DP-1 (pin 4) received hotplug event.
[2003687.986226] [drm:intel_hdmi_detect], [CONNECTOR:18:HDMI-A-1]
[2003687.988663] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus dpb
[2003687.988667] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2003687.991165] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2003688.000430] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2003688.008441] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2003690.020909] [drm:intel_crtc_cursor_set], cursor off

[2003730.883208] [drm:drm_mode_getconnector], [CONNECTOR:9:?]
[2003730.883225] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:9:LVDS-1]
[2003730.883234] [drm:intel_lvds_detect], [CONNECTOR:9:LVDS-1]
[2003730.883262] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:9:LVDS-1] probed modes :
[2003730.883270] [drm:drm_mode_debug_printmodeline], Modeline 12:"1600x900" 60 108000 1600 1648 1680 1924 900 903 908 936 0x48 0xa
[2003730.883287] [drm:drm_mode_getconnector], [CONNECTOR:9:?]
[2003730.883748] [drm:drm_mode_getconnector], [CONNECTOR:15:?]
[2003730.883757] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:15:VGA-1]
[2003730.883765] [drm:intel_crt_detect], [CONNECTOR:15:VGA-1] force=1
[2003730.883777] [drm:intel_ironlake_crt_detect_hotplug], ironlake hotplug adpa=0xf40000, result 0
[2003730.883784] [drm:intel_crt_detect], CRT not detected via hotplug
[2003730.898945] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus vga
[2003730.898958] [drm:intel_crt_detect_ddc], CRT not detected via DDC:0x50 [no valid EDID found]
[2003730.898967] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:15:VGA-1] disconnected
[2003730.899077] [drm:drm_mode_getconnector], [CONNECTOR:18:?]
[2003730.899087] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:18:HDMI-A-1]
[2003730.899094] [drm:intel_hdmi_detect], [CONNECTOR:18:HDMI-A-1]
[2003730.899578] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus dpb
[2003730.899586] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:18:HDMI-A-1] disconnected
[2003730.899597] [drm:drm_mode_getconnector], [CONNECTOR:22:?]
[2003730.899604] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1]
[2003730.899611] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2003730.902160] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2003730.909465] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2003730.917464] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2003730.922974] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1] disconnected
```

shenki, 2015-01-16:
This is the firmware we are testing with:

https://jms.id.au/~joel/saturday-test.tar.xz

It is built from https://github.com/shenki/HDMI2USB/commit/ff4a1bde588168e23c66c7a98cce01262cf9edb0

shenki, 2015-01-16:
@mithro what did you turn on to enable those drm debug messages?

mithro, 2015-01-16:
See https://github.com/timvideos/HDMI2USB/issues/105#issuecomment-70193767 above.

mithro, 2015-01-16:
Plug in
```
[2010537.007099] [drm:intel_hpd_irq_handler], Received HPD interrupt on PIN 4 - cnt: 0
[2010537.007145] [drm:i915_hotplug_work_func], running encoder hotplug functions
[2010537.007153] [drm:i915_hotplug_work_func], Connector HDMI-A-1 (pin 4) received hotplug event.
[2010537.007157] [drm:i915_hotplug_work_func], Connector DP-1 (pin 4) received hotplug event.
[2010537.007162] [drm:intel_hdmi_detect], [CONNECTOR:18:HDMI-A-1]
[2010537.007707] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus dpb
[2010537.007711] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2010537.010386] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2010537.019681] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2010537.027649] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2010539.675729] [drm:intel_crtc_cursor_set], cursor off
```

----
```
[2010598.652928] [drm:intel_crtc_cursor_set], cursor off
[2010607.211568] [drm:drm_mode_getconnector], [CONNECTOR:9:?]
[2010607.211574] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:9:LVDS-1]
[2010607.211575] [drm:intel_lvds_detect], [CONNECTOR:9:LVDS-1]
[2010607.211584] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:9:LVDS-1] probed modes :
[2010607.211586] [drm:drm_mode_debug_printmodeline], Modeline 12:"1600x900" 60 108000 1600 1648 1680 1924 900 903 908 936 0x48 0xa
[2010607.211589] [drm:drm_mode_getconnector], [CONNECTOR:9:?]
[2010607.211693] [drm:drm_mode_getconnector], [CONNECTOR:15:?]
[2010607.211694] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:15:VGA-1]
[2010607.211696] [drm:intel_crt_detect], [CONNECTOR:15:VGA-1] force=1
[2010607.211701] [drm:intel_ironlake_crt_detect_hotplug], ironlake hotplug adpa=0xf40000, result 0
[2010607.211702] [drm:intel_crt_detect], CRT not detected via hotplug
[2010607.225276] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus vga
[2010607.225287] [drm:intel_crt_detect_ddc], CRT not detected via DDC:0x50 [no valid EDID found]
[2010607.225294] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:15:VGA-1] disconnected
[2010607.225364] [drm:drm_mode_getconnector], [CONNECTOR:18:?]
[2010607.225369] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:18:HDMI-A-1]
[2010607.225371] [drm:intel_hdmi_detect], [CONNECTOR:18:HDMI-A-1]
[2010607.310445] [drm:drm_edid_to_eld], ELD: no CEA Extension found
[2010607.310449] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:18:HDMI-A-1] probed modes :
[2010607.310450] [drm:drm_mode_debug_printmodeline], Modeline 28:"1024x768" 60 65000 1024 1048 1184 1344 768 771 777 806 0x48 0xa
[2010607.310475] [drm:drm_mode_getconnector], [CONNECTOR:18:?]
[2010607.312620] [drm:drm_mode_getconnector], [CONNECTOR:22:?]
[2010607.312633] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1]
[2010607.312638] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2010607.315157] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2010607.323884] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2010607.331902] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2010607.337413] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1] disconnected
[2010607.348400] [drm:intel_crtc_cursor_set], cursor off
[2010607.353198] [drm:drm_mode_addfb], [FB:29]
[2010607.355894] [drm:drm_mode_setcrtc], [CRTC:3]
[2010607.355905] [drm:drm_mode_setcrtc], [CONNECTOR:9:LVDS-1]
[2010607.355908] [drm:intel_crtc_set_config], [CRTC:3] [FB:29] #connectors=1 (x y) (0 0)
[2010607.355912] [drm:intel_set_config_compute_mode_changes], computed changes for [CRTC:3], mode_changed=0, fb_changed=1
[2010607.355915] [drm:intel_modeset_stage_output_state], [CONNECTOR:9:LVDS-1] to [CRTC:3]
[2010607.363175] [drm:ironlake_update_plane], Writing base 0C17C000 00000000 0 0 10752
[2010607.386488] [drm:drm_mode_setcrtc], [CRTC:5]
[2010607.386506] [drm:drm_mode_setcrtc], [CONNECTOR:18:HDMI-A-1]
[2010607.386510] [drm:intel_crtc_set_config], [CRTC:5] [FB:29] #connectors=1 (x y) (1600 0)
[2010607.386514] [drm:intel_set_config_compute_mode_changes], inactive crtc, full mode set
[2010607.386515] [drm:intel_set_config_compute_mode_changes], modes are different, full mode set
[2010607.386517] [drm:drm_mode_debug_printmodeline], Modeline 51:"" 0 83500 1280 1352 1480 1680 800 803 809 831 0x0 0x5
[2010607.386520] [drm:drm_mode_debug_printmodeline], Modeline 30:"" 0 65000 1024 1048 1184 1344 768 771 777 806 0x0 0xa
[2010607.386523] [drm:intel_set_config_compute_mode_changes], computed changes for [CRTC:5], mode_changed=1, fb_changed=0
[2010607.386525] [drm:intel_modeset_stage_output_state], encoder changed, full mode switch
[2010607.386527] [drm:intel_modeset_stage_output_state], [CONNECTOR:9:LVDS-1] to [CRTC:3]
[2010607.386528] [drm:intel_modeset_stage_output_state], [CONNECTOR:18:HDMI-A-1] to [CRTC:5]
[2010607.386530] [drm:intel_modeset_stage_output_state], crtc changed, full mode switch
[2010607.386532] [drm:intel_modeset_affected_pipes], set mode pipe masks: modeset: 2, prepare: 2, disable: 0
[2010607.386535] [drm:connected_sink_compute_bpp], [CONNECTOR:18:HDMI-A-1] checking for sink bpp constrains
[2010607.386537] [drm:intel_hdmi_compute_config], picking bpc to 8 for HDMI output
[2010607.386538] [drm:intel_hdmi_compute_config], forcing pipe bpc to 24 for HDMI
[... 54 lines cut ...]
[2010607.693012] [drm:drm_mode_getconnector], [CONNECTOR:9:?]
[2010607.693126] [drm:drm_mode_getconnector], [CONNECTOR:15:?]
[... cut ...]
[2010607.792427] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1]
[2010607.792432] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2010607.794980] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2010607.804438] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2010607.812462] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2010607.817922] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1] disconnected
[2010618.181766] [drm:intel_crtc_cursor_set], cursor off
```

mithro, 2015-01-16:
A bunch of logs uploaded from my machine to https://drive.google.com/file/d/0B5VlNZ_Rvdw6Skk4bU15OFRpTXM/view?usp=sharing

mithro, 2015-01-16:
What happens when I disconnected the connector;

```
[2012178.057241] [drm:intel_hpd_irq_handler], Received HPD interrupt on PIN 4 - cnt: 0
[2012178.057296] [drm:i915_hotplug_work_func], running encoder hotplug functions
[2012178.057304] [drm:i915_hotplug_work_func], Connector HDMI-A-1 (pin 4) received hotplug event.
[2012178.057309] [drm:i915_hotplug_work_func], Connector DP-1 (pin 4) received hotplug event.
[2012178.057328] [drm:intel_hdmi_detect], [CONNECTOR:18:HDMI-A-1]
[2012178.072997] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus dpb
[2012178.073000] [drm:intel_hpd_irq_event], [CONNECTOR:18:HDMI-A-1] status updated from connected to disconnected
[2012178.073002] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2012178.074463] [drm:drm_mode_getconnector], [CONNECTOR:9:?]
[2012178.074468] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:9:LVDS-1]
[2012178.074470] [drm:intel_lvds_detect], [CONNECTOR:9:LVDS-1]
[2012178.074480] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:9:LVDS-1] probed modes :
[2012178.074483] [drm:drm_mode_debug_printmodeline], Modeline 12:"1600x900" 60 108000 1600 1648 1680 1924 900 903 908 936 0x48 0xa
[2012178.074488] [drm:drm_mode_getconnector], [CONNECTOR:9:?]
[2012178.074618] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2012178.074633] [drm:drm_mode_getconnector], [CONNECTOR:15:?]
[2012178.074636] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:15:VGA-1]
[2012178.074638] [drm:intel_crt_detect], [CONNECTOR:15:VGA-1] force=1
[2012178.074644] [drm:intel_ironlake_crt_detect_hotplug], ironlake hotplug adpa=0xf40000, result 0
[2012178.074646] [drm:intel_crt_detect], CRT not detected via hotplug
[2012178.089027] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus vga
[2012178.089035] [drm:intel_crt_detect_ddc], CRT not detected via DDC:0x50 [no valid EDID found]
[2012178.089041] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:15:VGA-1] disconnected
[2012178.089065] [drm:intel_hdmi_detect], 
[2012178.089071] [drm:drm_mode_getconnector], [CONNECTOR:18:?]
[2012178.089080] [CONNECTOR:18:HDMI-A-1]
[2012178.105033] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus dpb
[2012178.105036] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:18:HDMI-A-1]
[2012178.105038] [drm:intel_hdmi_detect], [CONNECTOR:18:HDMI-A-1]
[2012178.121053] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus dpb
[2012178.121055] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:18:HDMI-A-1] disconnected
[2012178.121059] [drm:intel_lvds_detect], [CONNECTOR:9:LVDS-1]
[2012178.121083] [drm:intel_crt_detect], [CONNECTOR:15:VGA-1] force=1
[2012178.121088] [drm:intel_ironlake_crt_detect_hotplug], ironlake hotplug adpa=0xf40000, result 0
[2012178.121090] [drm:intel_crt_detect], CRT not detected via hotplug
[2012178.121576] [drm:drm_mode_getconnector], [CONNECTOR:22:?]
[2012178.137071] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus vga
[2012178.137073] [drm:intel_crt_detect_ddc], CRT not detected via DDC:0x50 [no valid EDID found]
[2012178.137111] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1]
[2012178.137114] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2012178.137120] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1] disconnected
[2012178.151812] [drm:drm_mode_setcrtc], [CRTC:5]
[2012178.151826] [drm:intel_crtc_set_config], [CRTC:5] [NOFB]
[2012178.151834] [drm:intel_set_config_compute_mode_changes], computed changes for [CRTC:5], mode_changed=1, fb_changed=0
[2012178.151841] [drm:intel_modeset_stage_output_state], [CONNECTOR:18:HDMI-A-1] to [NOCRTC]
[2012178.151845] [drm:intel_modeset_stage_output_state], encoder changed, full mode switch
[2012178.151849] [drm:intel_modeset_stage_output_state], [CONNECTOR:9:LVDS-1] to [CRTC:3]
[2012178.151854] [drm:intel_modeset_stage_output_state], crtc changed, full mode switch
[2012178.151859] [drm:intel_modeset_affected_pipes], set mode pipe masks: modeset: 0, prepare: 0, disable: 2
[2012178.205180] [drm:intel_disable_shared_dpll], disable PCH DPLL B (active 1, on? 1) for crtc 5
[2012178.205198] [drm:intel_disable_shared_dpll], disabling PCH DPLL B
[2012178.205619] [drm:ivybridge_update_wm], FIFO watermarks For pipe A - plane 11, cursor: 6
[2012178.205625] [drm:ironlake_check_srwm], watermark 1: display plane 16, fbc lines 3, cursor 6
[2012178.205630] [drm:ironlake_check_srwm], watermark 2: display plane 56, fbc lines 3, cursor 6
[2012178.205635] [drm:ironlake_check_srwm], watermark 3: display plane 102, fbc lines 4, cursor 6
[2012178.205640] [drm:ironlake_check_srwm], watermark 3: display plane 202, fbc lines 5, cursor 10
[2012178.205647] [drm:intel_update_fbc], disabled per chip default
[2012178.205666] [drm:ivb_modeset_global_resources], disabling fdi C rx
[2012178.205675] [drm:intel_connector_check_state], [CONNECTOR:9:LVDS-1]
[2012178.205689] [drm:check_encoder_state], [ENCODER:10:LVDS-10]
[2012178.205698] [drm:check_encoder_state], [ENCODER:16:DAC-16]
[2012178.205706] [drm:check_encoder_state], [ENCODER:17:TMDS-17]
[2012178.205714] [drm:check_encoder_state], [ENCODER:21:TMDS-21]
[2012178.205720] [drm:check_crtc_state], [CRTC:3]
[2012178.205746] [drm:check_crtc_state], [CRTC:5]
[2012178.205750] [drm:check_crtc_state], [CRTC:7]
[2012178.205754] [drm:check_shared_dpll_state], PCH DPLL A
[... 8 lines cut ...]
[2012178.246687] [drm:intel_crtc_cursor_set], cursor off
[... cut ...]
[2012178.337310] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus dpb
[2012178.337312] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:18:HDMI-A-1] disconnected
[2012178.337328] [drm:drm_mode_getconnector], [CONNECTOR:22:?]
[2012178.337330] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1]
[2012178.337331] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2012178.337337] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1] disconnected
```

mithro, 2015-01-16:
Plug fail
```
[2012297.018331] [drm:intel_hpd_irq_handler], Received HPD interrupt on PIN 4 - cnt: 0
[2012297.018393] [drm:i915_hotplug_work_func], running encoder hotplug functions
[2012297.018402] [drm:i915_hotplug_work_func], Connector HDMI-A-1 (pin 4) received hotplug event.
[2012297.018418] [drm:i915_hotplug_work_func], Connector DP-1 (pin 4) received hotplug event.
[2012297.018420] [drm:intel_hdmi_detect], [CONNECTOR:18:HDMI-A-1]
[2012297.020201] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus dpb
[2012297.020204] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2012297.022680] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012297.031236] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012297.039124] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
```

mithro, 2015-01-16:
More interesting plug;

```
[2012385.593835] [drm:intel_hpd_irq_handler], Received HPD interrupt on PIN 4 - cnt: 0
[2012385.593851] [drm:i915_hotplug_work_func], running encoder hotplug functions
[2012385.593854] [drm:i915_hotplug_work_func], Connector HDMI-A-1 (pin 4) received hotplug event.
[2012385.593855] [drm:i915_hotplug_work_func], Connector DP-1 (pin 4) received hotplug event.
[2012385.593857] [drm:intel_hdmi_detect], [CONNECTOR:18:HDMI-A-1]
[2012385.594558] i2c i2c-4: sendbytes: NAK bailout.
[2012385.637363] [drm:intel_hpd_irq_event], [CONNECTOR:18:HDMI-A-1] status updated from disconnected to connected
[2012385.637368] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2012385.639893] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012385.648253] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012385.656264] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012385.664412] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2012385.666918] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012385.676291] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012385.684320] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012385.689856] [drm:drm_mode_getconnector], [CONNECTOR:9:?]
[2012385.689881] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:9:LVDS-1]
[2012385.689883] [drm:intel_lvds_detect], [CONNECTOR:9:LVDS-1]
[2012385.689893] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:9:LVDS-1] probed modes :
[2012385.689895] [drm:drm_mode_debug_printmodeline], Modeline 12:"1600x900" 60 108000 1600 1648 1680 1924 900 903 908 936 0x48 0xa
[2012385.689900] [drm:drm_mode_getconnector], [CONNECTOR:9:?]
[2012385.689905] [drm:intel_hdmi_detect], [CONNECTOR:18:HDMI-A-1]
[2012385.732604] [drm:intel_lvds_detect], [CONNECTOR:9:LVDS-1]
[2012385.732616] [drm:intel_crt_detect], [CONNECTOR:15:VGA-1] force=1
[2012385.732622] [drm:intel_ironlake_crt_detect_hotplug], ironlake hotplug adpa=0xf40000, result 0
[2012385.732623] [drm:intel_crt_detect], CRT not detected via hotplug
[2012385.732709] [drm:drm_mode_getconnector], [CONNECTOR:15:?]
[2012385.745856] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus vga
[2012385.745857] [drm:intel_crt_detect_ddc], CRT not detected via DDC:0x50 [no valid EDID found]
[2012385.745874] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:15:VGA-1]
[2012385.745877] [drm:intel_crt_detect], [CONNECTOR:15:VGA-1] force=1
[2012385.745882] [drm:intel_ironlake_crt_detect_hotplug], ironlake hotplug adpa=0xf40000, result 0
[2012385.745883] [drm:intel_crt_detect], CRT not detected via hotplug
[2012385.761889] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus vga
[2012385.761895] [drm:intel_crt_detect_ddc], CRT not detected via DDC:0x50 [no valid EDID found]
[2012385.761900] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:15:VGA-1] disconnected
[2012385.761931] [drm:drm_mode_getconnector], [CONNECTOR:18:?]
[2012385.761937] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:18:HDMI-A-1]
[2012385.761941] [drm:intel_hdmi_detect], [CONNECTOR:18:HDMI-A-1]
[2012385.764659] [drm:drm_do_probe_ddc_edid], drm: skipping non-existent adapter i915 gmbus dpb
[2012385.764664] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:18:HDMI-A-1] disconnected
[2012385.764671] [drm:drm_mode_getconnector], [CONNECTOR:22:?]
[2012385.764676] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1]
[2012385.764681] [drm:intel_dp_detect], [CONNECTOR:22:DP-1]
[2012385.767207] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012385.776363] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012385.784362] [drm:intel_dp_aux_ch], dp_aux_ch timeout status 0x7145003f
[2012385.789893] [drm:drm_helper_probe_single_connector_modes], [CONNECTOR:22:DP-1] disconnected
```

mithro, 2015-01-16:
@shenki 
I think the logs show that the system is reliably seeing the state change from the hot-plug pin.

shenki, 2015-01-16:
https://github.com/torvalds/linux/blob/2c623c11c7d87ada0121c5502358b30c9a849d2d/drivers/gpu/drm/i915/intel_dp.c#L904

```
        /* Timeouts occur when the device isn't connected, so they're
         * "normal" -- don't fill the kernel log with these */
        if (status & DP_AUX_CH_CTL_TIME_OUT_ERROR) {
                DRM_DEBUG_KMS("dp_aux_ch timeout status 0x%08x\n", status);
                ret = -ETIMEDOUT;
                goto out;
        }
```

mithro, 2015-01-16:
Found some more useful information at http://unix.stackexchange.com/questions/114359/how-to-get-edid-for-a-single-monitor

```
$ cat /sys/devices/pci0000:00/0000:00:02.0/drm/card0/card0-HDMI-A-1/edid | parse-edid 
Checksum Correct

Section "Monitor"
	Identifier "HDMI2USB"
	ModelName "HDMI2USB"
	VendorName "JAZ"
	# Monitor Manufactured week 6 of 2013
	# EDID version 1.3
	# Digital Display
	# Display Physical Size not given. Normal for projectors.
	Gamma 2.20
	Option "DPMS" "false"
	#Not giving standard mode: 1024x768, 60Hz
	Modeline 	"Mode 0" 65.00 1024 1048 1184 1344 768 771 777 806 -hsync -vsync 
EndSection
```

mithro, 2015-04-27:
Getting captures we can analyze later would be useful.

Setup instructions;
 * Make sure there are no jumpers on JP6 and JP7.
 * Connect the probe to JP6 (SCL) and JP7 (SDA). 
  * When these jumpers are connected, the EDID from the J3 (the HDMI output) is connected to the EDID lines of J2 (the HDMI input) which won't work with the FPGA firmware.
  * One side of these two jumpers will be connected to IC1 (a HDMI input buffer, computer -> FPGA) and the other side will be connected to IC2 (a HDMI output buffer, FPGA -> screen).
  * For looking at issues where the computer is not correctly detecting the FPGA as a monitor you want the side which is connected to IC1 side.
  * Load the firmware.
  * Connect the computer to HDMI connector J2.

This setup will only be testing the I2C on the "prebuffer" side of the system, which is 3V3. To test what is actually seen by computer you will need to either;
  * Create a special HDMI cable where you can access the I2C lines.
  * Solder breakout wires onto the HDMI connector pads.

The raw EDID file which is loaded into the FPGA memory can be found at https://github.com/timvideos/HDMI2USB/blob/master/hdl/edid/edidrom.hex or https://github.com/timvideos/HDMI2USB/blob/master/hdl/edid/hdmirom.hex

VHDL for the I2C slave in the FPGA can be found at 
https://github.com/timvideos/HDMI2USB/blob/master/hdl/edid/edidslave.v

mithro, 2015-04-28:
There has been some discussion on this issue on the mailing list at https://groups.google.com/forum/#!topic/hdmi2usb/AwyF-l0kslM

mithro, 2015-10-01:
This problem has been solved in the misoc based firmware at https://github.com/timvideos/HDMI2USB-misoc-firmware - we have found only one laptop which doesn't work reliably with that implementation so far.

mithro, 2016-01-10:
The new [HDMI2USB-misoc-firmware](https://github.com/timvideos/HDMI2USB-misoc-firmware) doesn't suffer from these problems.

closed 2016-01-10 by mithro