timvideos/HDMI2USB-jahanzeb-firmware #120: UVC non compliance - GET_DEF(PROBE) not supported

CarlFK, 2015-03-31:
so that v4l2-compliance doesn't error:

$ v4l2-compliance -d /dev/video1

		fail: v4l2-test-controls.cpp(513): g_ext_ctrls worked even when no controls are present
	test VIDIOC_G/S/TRY_EXT_CTRLS: FAIL

And we don't get the note in dmesg:

```
[ 5188.310191] usb 2-1.3: new high-speed USB device number 7 using ehci-pci
[ 5188.403409] usb 2-1.3: New USB device found, idVendor=1d50, idProduct=60b7
[ 5188.403417] usb 2-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 5188.403422] usb 2-1.3: Product: HDMI2USB
[ 5188.403425] usb 2-1.3: Manufacturer: JANI
[ 5188.403900] uvcvideo: Found UVC 1.00 device HDMI2USB (1d50:60b7)
[ 5188.404196] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
```

 https://github.com/timvideos/HDMI2USB/blob/master/cypress/uvc.c#L94 

                 // case UVC_GET_DEF:
                 // FIXME: Missing this case causes the following errors
                 // uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.

mithro, 2016-01-10:
This issue was moved to timvideos/HDMI2USB-misoc-firmware#148

closed 2016-01-10 by mithro

Mahesh-dev-avula, 2023-03-13:
Hi @mithro ,
**Found UVC 1.10 device Devkit (0525:a4a7)
[18655674.612204] uvcvideo: UVC non compliance - GET_DEF(PROBE) not supported. Enabling workaround.
[18655674.836657] uvcvideo: Failed to query (129) UVC probe control : 26 (exp. 34).
[18655674.836662] uvcvideo: Failed to initialize the device (-5).**

How you have resolved this issue..?

could you please help on this.