timvideos/HDMI2USB-jahanzeb-firmware #128: Fix the pattern generation to support both 720p and 1024x768 resolution generation

mithro, 2015-06-03:
The code for the pattern generation can be found at https://github.com/timvideos/HDMI2USB/blob/master/hdl/misc/pattern.vhd

rohit91, 2015-07-20:
Do we need static compile-time support for enabling either XGA or 720p support using generics/parameters, or dynamic support for switching b/w either resolutions when the hardware is running?

Compile-time support using generics is really simple.
Dynamic switching b/w resolutions would need DCM_CLKGEN to be programmed on-the-fly using SPI for switching b/w 65MHz Pixel clock (for XGA) and 75MHz Pixel clock (for 720p). Also, We need to expose a configuration option to UART for switching b/w resolutions.

ajitmathew, 2015-07-20:
I guess dynamic switching support is required. Also, if we have BUFGMUX
left in our design, then we can use it to multiplex clocks instead of re
configuring DCM every time we switch.

mithro, 2016-01-10:
Closing this as the [HDMI2USB-jahanzeb-firmware](https://github.com/timvideos/HDMI2USB-jahanzeb-firmware) is no longer under development

closed 2016-01-10 by mithro