timvideos/HDMI2USB-jahanzeb-firmware #130: Support the Numato Opsis board

mithro, 2015-07-17:
The [Numato](http://numato.com) Opsis board is the first [HDMI2USB](http://hdmi2usb.tv)
production board developed in conjugation with the [TimVideos.us](http://code.timvideos.us) project. 

It's features include;
 * 2 x HDMI input ports,
 * 2 x HDMI output ports,
 * DisplayPort input and output ports,
 * 128Mbit DDR3-1600 ram,
 * Gigabit Ethernet with MAC address EEPROM,
 * Micro SD card,
 * Cypress FX2 for USB device functionality,
 * ULPI for USB OTG functionality,
 * Quad speed SPI flash.

Sample code for the Opsis can be found at https://github.com/mithro/numato-opsis-sample-code

vitecd, 2015-07-17:
Hi
Unfortunately I can't found any notes about 'opsis' at Numato Web site

mithro, 2015-07-19:
@vitecd - The opsis board is not yet available for purchase as we are still in the initial bring up stage (and already have discovered a couple of very small bugs with the design that we wish to fix before beginning full scale production).  We want to verify the firmware works with the board before announcing it officially.

The board schematic and PCB design *will* be publish shortly (maybe later today depending on how things go). Will keep you informed.

mithro, 2015-10-01:
The Numato Opsis board is now crowdfunding at https://www.crowdsupply.com/numato-lab/opsis
The board is also supported by the misoc based firmware - https://github.com/timvideos/HDMI2USB-misoc-firmware

mithro, 2015-10-01:
With the misoc firmware already being in a much better state - we won't be porting this firmware to the Opsis board.

closed 2015-10-01 by mithro